Xilinx/mlir-aie #2363: Stack and Memory banks in AIE-ML architecture

flxinxout, 2025-06-03:
Hello,

When creating kernels run by XRT, we map memory banks from a compute tile with a buffer on host side. But, I suppose that kernel's arguments are stored in some stack in the compute tile. So my question is how its handled? Does every banks have a kind of stack and heap? Is there a real one-to-one mapping between host side buffers and memory banks?

Assume that my kernel has 3 arguments, and XRT declares 3 group ids, one for each, is there a kind of "stack" on another memory banks for loading the data in the kernel? 

Thanks!

andrej, 2025-06-04:
Hi @flxinxout 

Thanks for your question and the interest in our repository. 

Each AI Engine core has its own stack and heap (scratch pad memory). Neighboring tiles can additionally access each other's memories.

In the NPU, you cannot _directly_ map host buffers to compute tile buffers. All data movement is explicit, so you must instead describe the movement of data from the host through the NPU towards your desired compute tile manually. This is different from GPU and CPU programming, so takes some getting used to, but is very powerful. This also means that to see the link between host-side buffers and buffers on the compute tile, you must follow the entire data movement path described in the designs.

Describing data movement consists of creating a flow for data from the _shim_ tiles to the _compute_ tiles, and creating DMA programs that "shovel" data from streams into the buffers you want. You can do this in different ways, but a common high-abstraction-level way to do this is using ObjectFIFOs. When you create an ObjectFIFO, one or multiple buffers on the compute tile are allocated, a flow is set up between the shim and the compute tile, and two DMA programs are created: One that moves data from the host-side buffers onto the stream (via the configured flow), and one that takes data coming in from the stream on the compute tile into the buffers.

The last puzzle piece that connects the host-side buffers with the ObjectFIFOs (or other means of moving data inside the NPU) is the _runtime sequence_. The runtime sequence is a program that gets executed on the dedicated command processor on the NPU whenever you invoke your design via XRT on the host side. The dedicated command processor has the capacity to (re-)configure the NPU and dispatch work to it. If you look at an example runtime sequence like this one from `programming_examples/basic/passthrough_dmas/passthrough_dmas.py`, you can see that it takes three arguments (in this case, each of `vector_ty`). These three arguments correspond to the arguments you pass on the host side when you invoke the design (minus the runtime sequence itself, the length of the runtime sequence, and an opcode, which are the first three arguments and are needed by the firmware running on the command processor).

```
rt = Runtime()
#  arguments:   (1)        (2)        (3)            (1)  (2) (3)
with rt.sequence(vector_ty, vector_ty, vector_ty) as (a_in, _, c_out):
    rt.fill(of_in.prod(), a_in)
    rt.drain(of_out.cons(), c_out, wait=True)
```

On the host, this is invoked as follows:

```
  unsigned int opcode = 3;
  // arguments                                        (1)     (2)     (3)
  auto run = kernel(opcode, bo_instr, instr_v.size(), bo_inA, bo_inB, bo_out);
  run.wait();
```

So, in summary, the link between host buffer and compute tile buffer is indirect, and you can see this link by looking at the ObjectFIFOs and the runtime sequence. You have complete control over how host buffers get mapped to compute tile buffers.

Hope this helps. Feel free to follow up if you have questions on how a specific example or even your code works (or why it doesn't work).

Best,
André

flxinxout, 2025-06-04:
Hello @andrej, thanks so much for the detailled answer. I still want to follow the discussion if it's alright for you.

1. With what you said, it means that you don't actually map "the buffer (1)" to the "bank (1)" in the NPU's buffer right? Its more related to the ObjectFIFOs I declare in my IRON file. The data that is actually sent to the NPU's internal memory is mapped somewhere in it, in the "stack" I suppose then? Do I have to think about that, I mean in terms of stack's size?

2. if I understood well also, because compute tiles have only 2 DMA channels in and 2 DMA channels out, you can only have 2 ObjectFIFOs going in and out of the compute tile right? 

3. why are we limited to 8 buffers on the host side? I remarked that by opening the xclbin file, I saw only 8 banks available, which meant for me initially that I was related to the "8 banks" of the compute tile.

I'm actually writing a report on 6 month leaning how to use MLIR-AIE, so I would like to have right answers to these type of question you understand haha! 

thanks!
Giovanni

andrej, 2025-06-09:
Yes, I believe what you linked is all the official public documentation that is available for the AI Engine intrinsics.

closed 2025-06-10 by andrej