Xilinx/mlir-aie #2404: How to compile kernel into my IRON py code

derekharker, 2025-06-18:
I've tried a few different ways on my own to include the .cc kernel files under aie_kernels/generic/ so that I can use them in my MLIR code, but I can't get any of them to work. How do I compile them into my host.exe? Or do I need to bring the file into my working directory

andrej, 2025-06-23:
Hi @derekharker, thanks for your interest and opening an issue.

You can see how we link C kernels into a design in our programming examples. The steps are 
1. [Compile the C kernel](https://github.com/Xilinx/mlir-aie/blob/2cf6582df15d38ac491407ff2826bda153656e43/programming_examples/basic/passthrough_kernel/Makefile#L35) into an object file (using the open-source peano llvm-aie compiler or chess compiler).
2. [Reference the object file](https://github.com/Xilinx/mlir-aie/blob/2cf6582df15d38ac491407ff2826bda153656e43/programming_examples/basic/passthrough_kernel/passthrough_kernel.py#L32) from the design in Python. Make sure your signature matches.
3. [Link everything together](https://github.com/Xilinx/mlir-aie/blob/2cf6582df15d38ac491407ff2826bda153656e43/programming_examples/basic/passthrough_kernel/Makefile#L53).

derekharker, 2025-07-07:
@andrej 

I'm trying to figure out how to use chess compiler. Would you recommend using xchesscc or aiecompiler?

Are there any examples?

jgmelber, 2025-07-07:
Hi @derekharker, xchesscc can be used to compile the .cc kernels, Peano from llvm-aie is recommended for AIE2 and later architectures. 

Examples of using xchesscc can be found in the repo: https://github.com/Xilinx/mlir-aie/blob/d7dc96822ba5916de613dc1c06aa7d3116c83f3e/programming_examples/basic/vector_scalar_mul/Makefile#L45-L46

jgmelber, 2025-07-08:
`/home/dwharker/mlir-aie/llvm/install/bin/clang++` I think that might be clang built from llvm source rather than llvm-aie Peano compiler's clang

jgmelber, 2025-07-09:
That builds LLVM from source in order to build the mlir-aie repo, you can also use a[ pre-built llvm wheel](https://github.com/Xilinx/mlir-aie/blob/main/utils/build-mlir-aie-from-wheels.sh). 

```
# Install Peano from llvm-aie wheel
python3 -m pip install llvm-aie -f https://github.com/Xilinx/llvm-aie/releases/expanded_assets/nightly
export PEANO_INSTALL_DIR="$(pip show llvm-aie | grep ^Location: | awk '{print $2}')/llvm-aie"

# Peano clang can be called by: ${PEANO_INSTALL_DIR}/bin/clang++
```

https://github.com/Xilinx/mlir-aie/blob/8c5e34233df039c1c42ce427a8ca29ffb263c18b/programming_examples/basic/passthrough_kernel/Makefile#L37-L40

You aren't required to build Peano or LLVM from source to work with this project.

derekharker, 2025-07-11:
I'm using from source because it's required for it to work for the VCK 190 board that I'm using. I think the mlir tutorial 9 has a solution that is currently working right now. Wondering if it will work for my files, but I'll go with it for now. Using xchesscc and xchesscc_wrapper

closed 2025-07-17 by derekharker