Xilinx/mlir-aie #2406: aie2p Bitwise Intrinsics Support for int16_t

yenjames, 2025-06-18:
Hi team,

I am interested in creating an IRON placed version of [multi-head attention](https://github.com/Xilinx/mlir-air/tree/7f2d3198883c53c10cd2eb43a99cfbc897393d72/programming_examples/llama2_mha) from the **mlir-air** repo, using the available kernels from [mha.cc](https://github.com/Xilinx/mlir-air/blob/7f2d3198883c53c10cd2eb43a99cfbc897393d72/programming_examples/llama2_mha/mha.cc). However, I think some intrinsics are not available for npu2:

- [ `::shuffle_T16_8x2`:](https://github.com/Xilinx/mlir-air/blob/7f2d3198883c53c10cd2eb43a99cfbc897393d72/programming_examples/llama2_mha/mha.cc#L228) I believe this just interleaves two groups of vectors of 8 elements, and I can do this manually element-wise.
- [Quadrant determination in `sinf_cosf_poly_bf16` with bitwise ops:](https://github.com/Xilinx/mlir-air/blob/7f2d3198883c53c10cd2eb43a99cfbc897393d72/programming_examples/llama2_mha/mha.cc#L140-L150). I think the issue comes from:
```
    aie::mask<n> is_odd, is_neg;
    if (isSine) {
      is_odd = eq(bit_and(z, oneint16), oneint16); // select sin or cos
      is_neg =
          eq(bit_and(z, twoint16), twoint16); // select positive or negetive
    } else {
      is_odd = eq(bit_and(z, oneint16), zeroint16); // select sin or cos
      is_neg =
          eq(bit_xor(downshift(bit_and(z, twoint16), 1), bit_and(z, oneint16)),
             oneint16); // select positive or negetive
    }
```
Using `--target=aie2p-none-unknown-elf` for Peano:
```
fatal error: error in backend: unable to legalize instruction: %140:_(<4 x s32>) = G_AND %78:_, %138:_ (in function: _Z19sinf_cosf_poly_bf16ILj24ELj8ELb1EEvPK8bfloat16PS0_)
```
I suspect this may be due to the lack of support for bitwise operations on int16_t vectors in the Strix Point (aie2p) backend. 

Aside, would I also need a npu2 version of [lut-based softmax ops](https://github.com/Xilinx/mlir-air/blob/7f2d3198883c53c10cd2eb43a99cfbc897393d72/programming_examples/llama2_mha/lut_based_ops.h)? Thank you!

erwei-xilinx, 2025-06-20:
Hi @yenjames, 

Thanks for trying out the mlir-air example with aie2p. I don't know if any aie2 support for shuffle or bitwise operations have been removed in aie2p, but I suspect that their names (e.g. shuffle mode names) and data width assumptions have changed, since the data interface in aie2p has been widened from 256 to 512 bits.

I couldn't find a public documentation detailing the aie2p shuffle or bitwise ops, but I found this example (https://github.com/Xilinx/aie_api/blob/dc3ab77cd64c6ace5c073edd141d749b634031a4/include/aie_api/detail/aie2p/mmul_bfp16_bfp16.hpp#L55). Hopefully this repository could give you some more information on how to update a design from aie2 to aie2p.

yenjames, 2025-06-21:
Hi @erwei-xilinx,

Thank you for the response. I was able to resolve the shuffle issue, looks like that is due to a name change from `shuffle_T16_8x2` to `T16_8x2`.

Regarding the Peano codegen error:

- I’d like to retract my earlier assumption about it being related to unsupported bitwise intrinsics.
- The crash appears to stem from assigning a mask from a bitwise expression with a more complex variable like `z`
```cpp
aie::mask<n> is_odd;
is_odd = eq(bit_and(z, oneint16), oneint16);
```
If I remove the [is_odd / is_neg](https://github.com/Xilinx/mlir-air/blob/d9c3bafef4d54d752720a5a6edc6f5dff8a80631/programming_examples/llama2_mha/mha.cc#L140-L150) assignments entirely, the kernel compiles. Replacing [`z`](https://github.com/Xilinx/mlir-air/blob/d9c3bafef4d54d752720a5a6edc6f5dff8a80631/programming_examples/llama2_mha/mha.cc#L137) with a statically known vector also works. I tried reproducing with minimal testcase but the crash seems to happen only with the full `sinf_cosf_poly_bf16` kernel context on _AIE2P_. 

I understand this `mha.cc` kernel is not officially in mlir-aie, so no worries if there are other more crucial bugs. Thanks again!

jgmelber, 2025-07-09:
FYI @Victor-Jung

closed 2025-09-05 by AndraBisca