Xilinx/mlir-aie #2426: core elf file gets to large in convert-scf-to-cf mlir pass

z4m1n0, 2025-07-09:
Helllo all,

I started with the example [martix multiplication - whole array](https://github.com/Xilinx/mlir-aie/tree/main/programming_examples/basic/matrix_multiplication/whole_array) and adapted it to handle cases where the tile count for m and n is not evenly divisible by the number of rows and columns of the npu.

The problem now is, that some elf files get larger than 16KB and the compiler crashes.
```
cd build && aiecc.py --alloc-scheme=basic-sequential --aie-generate-xclbin --no-compile-host --xclbin-name=final.xclbin \
                         --no-xchesscc --no-xbridge --peano /home/tamino/ryzen_ai/mlir-aie/ironenv/lib/python3.12/site-packages/llvm-aie \
                        --aie-generate-npu-insts --npu-insts-name=insts.txt ../build/aie.mlir
Found xchesscc at /tools/Xilinx/Vitis/2024.2/aietools
 AIE Compilation: ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━╸━━━━━━━  82% 0:00:02 0:00:08 14/17 4 WorkersGenerating: /home/tamino/ryzen_ai/tensor-aie/build/aie.mlir.prj/aie_cdo_elfs.bin
[AIE ERROR] _XAie_LoadProgMemSection():231: Overflow of program memory

```

<details>
<summary>For example, the mlir code for the compute tiles looks like this: (34 lines)</summary>

```llvm
      %core_0_2 = aie.core(%tile_0_2) {
      %c0 = arith.constant 0 : index
      %c4294967295 = arith.constant 4294967295 : index
      %c1 = arith.constant 1 : index
      scf.for %arg0 = %c0 to %c4294967295 step %c1 {
        %c0_0 = arith.constant 0 : index
        %c39 = arith.constant 39 : index            // m-dimension
        %c1_1 = arith.constant 1 : index
        scf.for %arg1 = %c0_0 to %c39 step %c1_1 {
          %c0_2 = arith.constant 0 : index
          %c47 = arith.constant 47 : index          // n-dimension
          %c1_3 = arith.constant 1 : index
          scf.for %arg2 = %c0_2 to %c47 step %c1_3 {
            %0 = aie.objectfifo.acquire @OUT_L1L2_0_0(Produce, 1) : !aie.objectfifosubview<memref<48x40xbf16>>
            %1 = aie.objectfifo.subview.access %0[0] : !aie.objectfifosubview<memref<48x40xbf16>> -> memref<48x40xbf16>
            func.call @zero_bf16(%1) : (memref<48x40xbf16>) -> ()
            %c0_4 = arith.constant 0 : index
            %c151 = arith.constant 151 : index      // k-dimension
            %c1_5 = arith.constant 1 : index
            scf.for %arg3 = %c0_4 to %c151 step %c1_5 {
              %2 = aie.objectfifo.acquire @IN0_L2L1_0(Consume, 1) : !aie.objectfifosubview<memref<48x48xbf16>>
              %3 = aie.objectfifo.subview.access %2[0] : !aie.objectfifosubview<memref<48x48xbf16>> -> memref<48x48xbf16>
              %4 = aie.objectfifo.acquire @IN1_L2L1_0(Consume, 1) : !aie.objectfifosubview<memref<48x40xbf16>>
              %5 = aie.objectfifo.subview.access %4[0] : !aie.objectfifosubview<memref<48x40xbf16>> -> memref<48x40xbf16>
              func.call @matmul_bf16_bf16(%3, %5, %1) : (memref<48x48xbf16>, memref<48x40xbf16>, memref<48x40xbf16>) -> ()
              aie.objectfifo.release @IN0_L2L1_0(Consume, 1)
              aie.objectfifo.release @IN1_L2L1_0(Consume, 1)
            }
            aie.objectfifo.release @OUT_L1L2_0_0(Produce, 1)
          }
        }
      }
      aie.end
    } {link_with = "mm_bf16_48x48x40.o"}
```
</details>

After the lowering from scf to cf [1201 lines](https://pastebin.com/N9KUKaK8) are produced, which are compiled into a 26KB elf file. (same when n-dimension is 45)
```
26K Jul  9 12:49 core_0_2.elf
```

<details>
<summary>When the loop size of the n-dimension is changed from 47 to 48 (same by 46):</summary>

```llvm
    %core_0_2 = aie.core(%tile_0_2) {
      %c0 = arith.constant 0 : index
      %c4294967295 = arith.constant 4294967295 : index
      %c1 = arith.constant 1 : index
      scf.for %arg0 = %c0 to %c4294967295 step %c1 {
[... 151 lines cut ...]
    llvm.intr.assume %6 ["align"(%46, %5 : !llvm.ptr, i64)] : i1
    llvm.call @matmul_bf16_bf16(%46, %45, %39) : (!llvm.ptr, !llvm.ptr, !llvm.ptr) -> ()
    llvm.call @llvm.aie2.release(%10, %14) : (i32, i32) -> ()
    llvm.call @llvm.aie2.release(%9, %14) : (i32, i32) -> ()
    llvm.call @llvm.aie2.release(%8, %14) : (i32, i32) -> ()
    %47 = llvm.add %27, %17 : i64
    llvm.br ^bb3(%47 : i64)
  ^bb11:  // pred: ^bb3
    %48 = llvm.add %25, %22 : i64
    llvm.br ^bb2(%48 : i64)
  ^bb12:  // pred: ^bb2
    %49 = llvm.add %23, %22 : i64
    llvm.br ^bb1(%49 : i64)
  ^bb13:  // pred: ^bb1
    llvm.return
  }
```
</details>

and they are compiled into an 11KB elf file.
```
11K Jul  9 13:12 core_0_2.elf
```

This suggests to me that there is no inherent problem with the mlir that is generated and that the problem lies in the lowering pass, particularly when the n-loop size is odd and the k-loop size is greater than 1.

andrej, 2025-07-21:
I suspect this may be due to the ObjectFIFO unrolling. Try passing `--dynamic-objFifos` to `aiecc.py` during compilation to disable this unrolling.

z4m1n0, 2025-07-22:
Hello @andrej,

The flag `--dynamic-objFifos` resolved the current issue at hand and the lowered mlir is now 177 lines.

<details>
<summary>
  lowered code with `--dynamic-objFifos`
</summary>

```mlir
  llvm.func @core_0_2() {
    %0 = llvm.mlir.addressof @IN1_L2L1_0_0_cons_buff_1 : !llvm.ptr
    %1 = llvm.mlir.addressof @IN1_L2L1_0_0_cons_buff_0 : !llvm.ptr
    %2 = llvm.mlir.addressof @IN0_L2L1_0_0_cons_buff_1 : !llvm.ptr
    %3 = llvm.mlir.addressof @IN0_L2L1_0_0_cons_buff_0 : !llvm.ptr
    %4 = llvm.mlir.addressof @OUT_L1L2_0_0_buff_1 : !llvm.ptr
    %5 = llvm.mlir.poison : !llvm.struct<(ptr, ptr, i64, array<2 x i64>, array<2 x i64>)>
    %6 = llvm.mlir.addressof @OUT_L1L2_0_0_buff_0 : !llvm.ptr
    %7 = llvm.mlir.constant(40 : index) : i64
    %8 = llvm.mlir.constant(48 : index) : i64
    %9 = llvm.mlir.constant(3735928559 : index) : i64
    %10 = llvm.mlir.addressof @_anonymous0 : !llvm.ptr
    %11 = llvm.mlir.constant(53 : i32) : i32
    %12 = llvm.mlir.constant(50 : i32) : i32
    %13 = llvm.mlir.constant(48 : i32) : i32
    %14 = llvm.mlir.constant(51 : i32) : i32
    %15 = llvm.mlir.constant(49 : i32) : i32
    %16 = llvm.mlir.constant(52 : i32) : i32
    %17 = llvm.mlir.constant(1 : i32) : i32
    %18 = llvm.mlir.constant(151 : index) : i64
    %19 = llvm.mlir.constant(-1 : i32) : i32
    %20 = llvm.mlir.constant(47 : index) : i64
    %21 = llvm.mlir.constant(39 : index) : i64
    %22 = llvm.mlir.constant(4294967295 : index) : i64
    %23 = llvm.mlir.constant(1 : index) : i64
    %24 = llvm.mlir.constant(0 : i32) : i32
    %25 = llvm.mlir.constant(0 : index) : i64
    %26 = llvm.mlir.constant(2 : i32) : i32
    %27 = llvm.getelementptr %10[0, 0] : (!llvm.ptr) -> !llvm.ptr, !llvm.array<3 x i32>
    llvm.store %24, %27 : i32, !llvm.ptr
    %28 = llvm.getelementptr inbounds|nuw %27[1] : (!llvm.ptr) -> !llvm.ptr, i32
    llvm.store %24, %28 : i32, !llvm.ptr
    %29 = llvm.getelementptr inbounds|nuw %27[2] : (!llvm.ptr) -> !llvm.ptr, i32
    llvm.store %24, %29 : i32, !llvm.ptr
    llvm.br ^bb1(%25 : i64)
  ^bb1(%30: i64):  // 2 preds: ^bb0, ^bb18
    %31 = llvm.icmp "slt" %30, %22 : i64
    llvm.cond_br %31, ^bb2(%25 : i64), ^bb19
  ^bb2(%32: i64):  // 2 preds: ^bb1, ^bb17
    %33 = llvm.icmp "slt" %32, %21 : i64
    llvm.cond_br %33, ^bb3(%25 : i64), ^bb18
  ^bb3(%34: i64):  // 2 preds: ^bb2, ^bb16
    %35 = llvm.icmp "slt" %34, %20 : i64
    llvm.cond_br %35, ^bb4, ^bb17
  ^bb4:  // pred: ^bb3
    llvm.call @llvm.aie2.acquire(%16, %19) : (i32, i32) -> ()
    %36 = llvm.load %27 : !llvm.ptr -> i32
    llvm.switch %36 : i32, ^bb5 [
      0: ^bb5,
      1: ^bb6
    ]
  ^bb5:  // 2 preds: ^bb4, ^bb4
    %37 = llvm.getelementptr %6[0, 0, 0] : (!llvm.ptr) -> !llvm.ptr, !llvm.array<48 x array<40 x bf16>>
    %38 = llvm.inttoptr %9 : i64 to !llvm.ptr
    %39 = llvm.insertvalue %38, %5[0] : !llvm.struct<(ptr, ptr, i64, array<2 x i64>, array<2 x i64>)> 
    %40 = llvm.insertvalue %37, %39[1] : !llvm.struct<(ptr, ptr, i64, array<2 x i64>, array<2 x i64>)> 
    %41 = llvm.insertvalue %25, %40[2] : !llvm.struct<(ptr, ptr, i64, array<2 x i64>, array<2 x i64>)> 
    %42 = llvm.insertvalue %8, %41[3, 0] : !llvm.struct<(ptr, ptr, i64, array<2 x i64>, array<2 x i64>)> 
    %43 = llvm.insertvalue %7, %42[3, 1] : !llvm.struct<(ptr, ptr, i64, array<2 x i64>, array<2 x i64>)> 
    %44 = llvm.insertvalue %7, %43[4, 0] : !llvm.struct<(ptr, ptr, i64, array<2 x i64>, array<2 x i64>)> 
[... 96 lines cut ...]
    %111 = llvm.add %57, %23 : i64
    llvm.br ^bb8(%111 : i64)
  ^bb16:  // pred: ^bb8
    llvm.call @llvm.aie2.release(%11, %17) : (i32, i32) -> ()
    %112 = llvm.load %27 : !llvm.ptr -> i32
    %113 = llvm.add %112, %17 : i32
    %114 = llvm.icmp "sge" %113, %26 : i32
    %115 = llvm.add %112, %19 : i32
    %116 = llvm.select %114, %115, %113 : i1, i32
    llvm.store %116, %27 : i32, !llvm.ptr
    %117 = llvm.add %34, %23 : i64
    llvm.br ^bb3(%117 : i64)
  ^bb17:  // pred: ^bb3
    %118 = llvm.add %32, %23 : i64
    llvm.br ^bb2(%118 : i64)
  ^bb18:  // pred: ^bb2
    %119 = llvm.add %30, %23 : i64
    llvm.br ^bb1(%119 : i64)
  ^bb19:  // pred: ^bb1
    llvm.return
  }
```
</details>

I think there is a bug in either the unrolling of the ObjectFIFOs or in the lowering from scf. There is no reason why the lowered mlir should be 9 times larger when there is one fewer iteration.

andrej, 2025-07-22:
> I think there is a bug in either the unrolling of the ObjectFIFOs or in the lowering from scf. There is no reason why the lowered mlir should be 9 times larger when there is one fewer iteration.

This isn't necessarily a bug. I think it's entirely possible that unlucky iteration counts (i.e. nested loops with coprime iteration counts) can lead to a large unrolling. @AndraBisca Do you agree with this, or do you think there might be a bug here?

That being said, it may be useful for us to emit a warning for very high unroll counts and suggest enabling `--dynamic-objFifos` instead, or switching the default to be the dynamic ObjectFIFO altogether (my preference, unless this causes performance degradation).

closed 2025-07-23 by andrej