Xilinx/mlir-aie #2463: Seeking help: certain behaviours in core_body() causes Kernel returned ert_cmd_state.ERT_CMD_STATE_ABORT

RonZ13, 2025-07-31:
Hi there,

I was trying to create a CNN example using the placed level IRON. I found a weird behaviour with my program. The code below compiles, but it causes exceptions: Kernel returned ert_cmd_state.ERT_CMD_STATE_ABORT. In the **_core_body_** function, If I comment out the acquire&release of the objectFIFO that passes in the weight values and reduce the inner loop interations from 14 to 7. Then it runs without causing exceptions. I am puzzled with why this causes the kernal to crash. For example, the **edge_detect** example in the programing_examples, the **_core_body_** function that handles the filter2D can handle a inner loop with 1080 iterations. However, somehow for my case the inner loop cannot handle anything greater than 7 iterations. I might be misunderstanding some concepts here, so please feel free to correct me if I'm mistaken.

I have also attached my entire example files in a zip file at the bottom (Makefile + host program + kernel + placed-level IRON code that is shown here).

Your help is highly appreciated,
Ron

```
import numpy as np
import sys
import argparse

from aie.dialects.aie import *
from aie.dialects.aiex import *
from aie.helpers.dialects.ext.scf import _for as range_
from aie.extras.context import mlir_mod_ctx

import aie.utils.trace as trace_utils
from aie.utils.trace import PortEvent, ShimTilePortEvent
from aie.utils.trace_events_enum import CoreEvent, ShimTileEvent, MemTileEvent, MemEvent

dtype_map = {
    "i8": np.int8,
}

def CNN(opts):
    image_width = opts.image_width
    image_height = opts.image_height
    num_rows = opts.num_rows
    num_cols = opts.num_cols
    trace_size = opts.trace_size

    layer1_padding = 1
    layer1_stride = 1
    layer1_kernel_width = 3
    layer1_kernel_height = 3
    layer1_per_core_input_width = 16
    layer1_per_core_input_height = 16
    layer1_input_channel = 1
    layer1_per_core_output_width = 14
    layer1_per_core_output_height = 14
    layer1_output_channel = 8
    layer1_per_core_after_maxpool_width = 7
    layer1_per_core_after_maxpool_height = 7
    layer1_W_total = layer1_input_channel * layer1_kernel_width * layer1_kernel_height * layer1_output_channel
    layer1_X_total = layer1_input_channel * image_width * image_height
    layer1_Y_total = layer1_output_channel * layer1_per_core_output_width * layer1_per_core_output_height

    input_W_shape = (layer1_W_total,)
    input_X_shape = (layer1_X_total,)
    Output_shape = (layer1_Y_total,)
    dtype_str = opts.dtype
    dtype_in_str = dtype_str
    dtype_out_str = dtype_str
    dtype_in = dtype_map[dtype_in_str]
    dtype_out = dtype_map[dtype_out_str]
    input_W_ty = np.ndarray[input_W_shape, np.dtype[dtype_in]] 
    input_X_ty = np.ndarray[input_X_shape, np.dtype[dtype_in]] 
    output_ty = np.ndarray[Output_shape, np.dtype[dtype_out]] 

    layer1_W_ty = np.ndarray[(layer1_W_total,), np.dtype[dtype_in]]  # kernel weights for layer 1
    layer1_X_per_line_COMP_L1_ty = np.ndarray[(layer1_input_channel*layer1_per_core_input_width,), np.dtype[dtype_in]]      # Input per line for layer 1
    layer1_Y_per_line_COMP_L1_ty = np.ndarray[(layer1_output_channel*layer1_per_core_output_width,), np.dtype[dtype_in]]    # Output per line for layer 1

    @device(AIEDevice.npu2)
    def device_body():
        # AIE Core Function declarations
        layer1_conv2d = external_func(
[... 220 lines cut ...]
        help="Data type for the AIE array (e.g., int8)",
    )
    argparser.add_argument(
        "-m",
        type=int,
        default=32
    )
    argparser.add_argument(
        "-k",
        type=int,
        default=32
    )
    argparser.add_argument(
        "-n",
        type=int,
        default=32
    )
    opts = argparser.parse_args(sys.argv[1:])
    
    with mlir_mod_ctx() as ctx:
        CNN(opts)
        print(ctx.module)
```

[benchmarks.zip](https://github.com/user-attachments/files/21538967/benchmarks.zip)

AndraBisca, 2025-08-06:
Hello @RonZ13 and thank you for signaling this issue.

Based on the behavior you describe, this error can be related to a few particularities in the ObjectFIFO lowering code. Namely, for loops in the core code are unrolled such that the statically unrolled loop accesses each object in a FIFO-like manner. To determine by how much the loop needs to be unrolled, the depth of the ObjectFIFO is used, and the loop bounds and step are adjusted. There may also be remainder iterations after the loop that are added when the loop iterations are not divisible by the depth. This process can sometimes lead to a program memory overflow in the core.

As a fix, could you try enabling the dynamic lowering for ObjectFIFOs? You can do so by adding the `--dynamic-objFifos` flag to your aiecc.py call in your Makefile. 

Please let us know if this helped!

jackl-xilinx, 2025-08-06:
@RonZ13 Can you share the edit to `CNN_placed.py` that doesn't throw the error?

RonZ13, 2025-08-07:
Hi @jackl-xilinx 

for the edited CNN_placed.py, in core_body(), I commented out the weight FIFO and reduced the inner loop to only 7 iterations as shown below

```
def core_body():
                for _ in range_(sys.maxsize):
                    # kernel weights
                    # conv2d_kernel_weight = inOF_layer1_W_L2L1s[0].acquire(ObjectFifoPort.Consume, 1)
                    for _ in range_(layer1_per_core_output_height-7):
                        # Read 3 input lines
                        input_lines = inOF_layer1_X_L2L1s[idx].acquire(ObjectFifoPort.Consume, 3)
                        # Write output
                        output_line = outOF_L1L2s[idx].acquire(ObjectFifoPort.Produce, 1)
                        # Zero the output line
                        layer1_zero(output_line, output_line.shape[0])

                        # Call conv2d + relu function
                        # layer1_conv2d(
                        #     input_lines[0], input_lines[1], input_lines[2], 
                        #     conv2d_kernel_weight, 
                        #     output_line,
                        #     layer1_per_core_input_width, layer1_per_core_input_height, 
                        #     layer1_input_channel, layer1_output_channel,
                        #     layer1_kernel_width, layer1_kernel_height
                        # )

                        inOF_layer1_X_L2L1s[idx].release(ObjectFifoPort.Consume, 1)
                        outOF_L1L2s[idx].release(ObjectFifoPort.Produce, 1)
                    inOF_layer1_X_L2L1s[idx].release(ObjectFifoPort.Consume, 2)
                    # inOF_layer1_W_L2L1s[0].release(ObjectFifoPort.Consume, 1)
```

RonZ13, 2025-08-07:
Hi @AndraBisca 

I have tried adding `--dynamic-objFifos` flag to the aiecc.py call, but it did not help. The kernel is still returning the same exception.