Xilinx/mlir-aie #2501: Unexpected `qds_device::wait()` error when computation latency is long

Fangtangtang, 2025-08-12:
Hi,

I have a program computing on a single compute tile. It takes two inputs, performs a large amount of computation, and produces one output.
I noticed that when the computation latency is long, the program reports the following error:

```
terminate called after throwing an instance of 'std::runtime_error'
  what():  qds_device::wait() unexpected command state
```

I suspect that this happens because the long computation latency is mistakenly detected as a deadlock by some mechanism.
Is this behavior expected?
If there are cases with long intermediate computation processes, what would be a reasonable way to handle them?

A minimal example:

- Take two input matrices and do matrix multiplications
- Add an extra loop to increase computation time

When the iteration count is small, the program runs normally.
When the iteration count is large (e.g., 16384), the above error occurs.

```mlir
module {
  aie.device(npu1_4col) {
    func.func private @fill_zeros_i16_32_32_vector(memref<32x32xi16>)
    func.func private @matmul_scalar_i16_i16(memref<32x64xi16>, memref<64x32xi16>, memref<32x32xi16>)
    %shim_noc_tile_0_0 = aie.tile(0, 0)
    %shim_noc_tile_1_0 = aie.tile(1, 0)
    %shim_noc_tile_2_0 = aie.tile(2, 0)
    %mem_tile_0_1 = aie.tile(0, 1)
    %mem_tile_1_1 = aie.tile(1, 1)
    %mem_tile_2_1 = aie.tile(2, 1)
    %tile_0_2 = aie.tile(0, 2)
    aie.objectfifo @fifo_0(%mem_tile_0_1, {%tile_0_2}, 2 : i32) : !aie.objectfifo<memref<32x64xi16>> 
    aie.objectfifo @fifo_1(%shim_noc_tile_0_0, {%mem_tile_0_1}, 2 : i32) : !aie.objectfifo<memref<1x1x32x64xi16>> 
    aie.objectfifo @fifo_2(%mem_tile_1_1, {%tile_0_2}, 2 : i32) : !aie.objectfifo<memref<64x32xi16>> 
    aie.objectfifo @fifo_3(%shim_noc_tile_1_0, {%mem_tile_1_1}, 2 : i32) : !aie.objectfifo<memref<1x1x64x32xi16>> 
    aie.objectfifo @fifo_4(%tile_0_2, {%mem_tile_2_1}, 2 : i32) : !aie.objectfifo<memref<32x32xi16>> 
    aie.objectfifo @fifo_5(%mem_tile_2_1, {%shim_noc_tile_2_0}, 2 : i32) : !aie.objectfifo<memref<1x1x32x32xi16>> 
    aie.objectfifo.link [@fifo_1] -> [@fifo_0]([] [])
    aie.objectfifo.link [@fifo_3] -> [@fifo_2]([] [])
    aie.objectfifo.link [@fifo_4] -> [@fifo_5]([] [])
    %buffer_0_2 = aie.buffer(%tile_0_2) : memref<32x32xi16> 
    %core_0_2 = aie.core(%tile_0_2) {
      %c16384 = arith.constant 16384 : index
      %c0 = arith.constant 0 : index
      %c1 = arith.constant 1 : index
      %c9223372036854775807 = arith.constant 9223372036854775807 : index
      scf.for %arg0 = %c0 to %c9223372036854775807 step %c1 {
        %0 = aie.objectfifo.acquire @fifo_0(Consume, 1) : !aie.objectfifosubview<memref<32x64xi16>>
        %1 = aie.objectfifo.subview.access %0[0] : !aie.objectfifosubview<memref<32x64xi16>> -> memref<32x64xi16>
        %2 = aie.objectfifo.acquire @fifo_2(Consume, 1) : !aie.objectfifosubview<memref<64x32xi16>>
        %3 = aie.objectfifo.subview.access %2[0] : !aie.objectfifosubview<memref<64x32xi16>> -> memref<64x32xi16>
        scf.for %arg1 = %c0 to %c16384 step %c1 {
          func.call @fill_zeros_i16_32_32_vector(%buffer_0_2) {lib = "fill_zeros_i16_32_32_vector"} : (memref<32x32xi16>) -> ()
          func.call @matmul_scalar_i16_i16(%1, %3, %buffer_0_2) {lib = "matmul_scalar_i16_i16"} : (memref<32x64xi16>, memref<64x32xi16>, memref<32x32xi16>) -> ()
        }
        %4 = aie.objectfifo.acquire @fifo_4(Produce, 1) : !aie.objectfifosubview<memref<32x32xi16>>
        %5 = aie.objectfifo.subview.access %4[0] : !aie.objectfifosubview<memref<32x32xi16>> -> memref<32x32xi16>
        func.call @fill_zeros_i16_32_32_vector(%5) {lib = "fill_zeros_i16_32_32_vector"} : (memref<32x32xi16>) -> ()
        func.call @matmul_scalar_i16_i16(%1, %3, %5) {lib = "matmul_scalar_i16_i16"} : (memref<32x64xi16>, memref<64x32xi16>, memref<32x32xi16>) -> ()
        aie.objectfifo.release @fifo_0(Consume, 1)
        aie.objectfifo.release @fifo_2(Consume, 1)
        aie.objectfifo.release @fifo_4(Produce, 1)
      }
      aie.end
    } {link_with = "external0.o"}
    aiex.runtime_sequence(%arg0: memref<2048xi16>, %arg1: memref<1024xi16>, %arg2: memref<2048xi16>) {
      aiex.npu.dma_memcpy_nd(%arg0[0, 0, 0, 0][1, 1, 32, 64][0, 2048, 64, 1]) {id = 0 : i64, issue_token = true, metadata = @fifo_1} : memref<2048xi16>
      aiex.npu.dma_memcpy_nd(%arg2[0, 0, 0, 0][1, 1, 64, 32][0, 0, 32, 1]) {id = 0 : i64, issue_token = true, metadata = @fifo_3} : memref<2048xi16>
      aiex.npu.dma_memcpy_nd(%arg1[0, 0, 0, 0][1, 1, 32, 32][0, 1024, 32, 1]) {id = 0 : i64, issue_token = true, metadata = @fifo_5} : memref<1024xi16>
      aiex.npu.dma_wait {symbol = @fifo_5}
      aie.end
    }
  }
}
```
Thanks!

fifield, 2025-08-12:
> I suspect that this happens because the long computation latency is mistakenly detected as a deadlock 
You are probably correct. Have you tried passing a timeout to the `run::wait()` call? I haven't tried it myself recently but I know there is [a timeout argument](https://github.com/Xilinx/XRT/blob/master/src/runtime_src/core/include/xrt/xrt_kernel.h#L305). If that doesn't work you might want to ask over at https://github.com/amd/xdna-driver, since this is fundamentally a driver feature.

Fangtangtang, 2025-08-12:
Thank you so much for the quick reply! 

However, changing the timeout for `run::wait()` doesn’t seem to help. From what I understand, this call waits on the CPU side for the NPU kernel to finish, and by default it blocks until the run completes. I tried passing a timeout argument, but got:

```
Kernel did not complete. Returned status: 8
```

which I think means the CPU didn’t get a completion signal from the NPU kernel before the timeout.

I suspect the `qds_device::wait()` issue I got is actually on the NPU side.

Thanks for your suggestion. I’ll raise another issue at `xdna-driver`.

erwei-xilinx, 2025-08-14:
What XDNA driver version are you using? We encountered a similar issue with a driver released about 3-4 weeks ago, the issue then disappeared when we updated the driver about 1-2 weeks ago, and we were able to run tests with high latency again.