Xilinx/mlir-aie #2510: Non-reproducible Builds Due to Non-deterministic MLIR/XCLBIN Generation

mawad-amd, 2025-08-17:
## Summary
The `add_one.py` script generates different MLIR output files and compiled binaries when run multiple times, even though the input code is identical. This non-deterministic behavior affects both the MLIR generation and the final compiled xclbin files, leading to non-reproducible builds.

## Steps to Reproduce
0. Save the python string to `add_one.py`
```py
import subprocess
import numpy as np
import aie.iron as iron
from aie.iron import ObjectFifo, Program, Runtime, Worker
from aie.iron.placers import SequentialPlacer
import sys

def my_add_one_kernel():
    input_dtype = np.uint8
    output_dtype = np.uint8

    # Sizes
    input_size = 1024
    output_size = input_size

    # Define tensor types
    line_size = 4
    line_type = np.ndarray[(line_size,), np.dtype[input_dtype]]
    vector_type = np.ndarray[(line_size,), np.dtype[output_dtype]]

    # Dataflow with ObjectFifos
    input_fifo = ObjectFifo(line_type, name="input")
    output_fifo = ObjectFifo(line_type, name="output")

    # Task for the core to perform
    def core_fn(input_fifo, output_fifo):
        output_elem = output_fifo.acquire(1)
        input_elem = input_fifo.acquire(1)
        for i in range(line_size):
            output_elem[i] = input_elem[i]
        input_fifo.release(1)
        output_fifo.release(1)

    # Create a worker to perform the task
    my_worker = Worker(
        core_fn,
        [input_fifo.cons(), output_fifo.prod()],
    )

    # Runtime operations to move data to/from the AIE-array
    rt = Runtime()
    with rt.sequence(vector_type, vector_type, vector_type) as (
        input_seq,
        output_seq,
        _,
    ):
        rt.start(my_worker)
        rt.fill(input_fifo.prod(), input_seq)
        rt.drain(output_fifo.cons(), output_seq, wait=True)

    # Place components (assign the resources on the device) and generate an MLIR module
    return Program(iron.get_current_device(), rt).resolve_program(SequentialPlacer())

if __name__ == "__main__":
    # Generate the MLIR module
    run_id = sys.argv[1]
    mlir_module = my_add_one_kernel()

    # Write to file
    mlir_file_name = f"add_one_{run_id}.mlir"
    xclbin_file_name = f"final_{run_id}.xclbin"
    insts_file_name = f"insts_{run_id}.bin"
    with open(mlir_file_name, "w") as f:
        print(mlir_module, file=f)

    print(f"MLIR module written to {mlir_file_name}")

    subprocess.run(
        [
            "aiecc.py",
            "--aie-generate-xclbin",
            "--aie-generate-npu-insts",
            "--no-compile-host",
        ]
        + [
            f"--xclbin-name={xclbin_file_name}",
            f"--npu-insts-name={insts_file_name}",
            mlir_file_name,
        ]
    )
    print(f"Saved {xclbin_file_name} and {insts_file_name}")

```

1. Run `python add_one.py 0` - generates `add_one_0.mlir`, `final_0.xclbin`, `insts_0.bin`
2. Run `python add_one.py 1` - generates `add_one_1.mlir`, `final_1.xclbin`, `insts_1.bin`. Run step two until you get different MLIR.

## Expected Behavior
Both runs should generate identical MLIR code and binaries since the Python source code is the same.

## Actual Behavior
The generated MLIR and binary files differ in the order of ObjectFifo declarations:

### MLIR Differences
**add_one_0.mlir:**
```mlir
aie.objectfifo @output(%tile_0_2, {%shim_noc_tile_0_0}, 2 : i32) : !aie.objectfifo<memref<4xui8>> 
aie.objectfifo @input(%shim_noc_tile_0_0, {%tile_0_2}, 2 : i32) : !aie.objectfifo<memref<4xui8>> 
```

**add_one_1.mlir:**
```mlir
aie.objectfifo @input(%shim_noc_tile_0_0, {%tile_0_2}, 2 : i32) : !aie.objectfifo<memref<4xui8>> 
aie.objectfifo @output(%tile_0_2, {%shim_noc_tile_0_0}, 2 : i32) : !aie.objectfifo<memref<4xui8>> 
```

### Binary Differences
- `final_0.xclbin` ≠ `final_1.xclbin` (Binary files differ)
- `insts_0.bin` = `insts_1.bin` (NPU instructions are identical)

**Hex Dump Comparison:**
```
final_0.xclbin:  e1 0c a2 68 00 00 00 00 54 24 00 00 00 00 00 00  e1 0c a2 68 00 00 00 00
final_1.xclbin:  f8 0c a2 68 00 00 00 00 54 24 00 00 00 00 00 00  f8 0c a2 68 00 00 00 00
                ^^ ^^                                                                  ^^ ^^

final_0.xclbin:  06 0c b4 90 29 86 ee bb  9f 71 1c cf 5f 78 7b 84
final_1.xclbin:  03 08 c4 5c 11 35 b8 9c  06 0f 86 8a 52 ed 0f 59
                ^^ ^^ ^^ ^^ ^^ ^^ ^^ ^^  ^^ ^^ ^^ ^^ ^^ ^^ ^^ ^^

final_0.xclbin:  19 40 03 38 3b 11 00 98  e3 01 00 d0 df ff 3b 11  00 18 f3 01 00 d0 ee ff
final_1.xclbin:  19 40 03 38 3b 11 00 9a  c3 01 00 d0 df ff 3b 11  00 18 d3 01 00 d0 ee ff
                                    ^^ ^^  ^^ ^^                    ^^ ^^
```

# Additional Notes
`aiecc.py` also produces non-deterministic XCLBINs:

```console
$ aiecc.py --aie-generate-xclbin --aie-generate-npu-insts --no-compile-host --xclbin-name=test_0.xclbin --npu-insts-name=test_0.bin add_one_0.mlir

Generating: /home/muhaawad/git/amd/iron/mlir-aie/programming_examples/basic/bug/add_one_0.mlir.prj/aie_cdo_elfs.bin
Generating: /home/muhaawad/git/amd/iron/mlir-aie/programming_examples/basic/bug/add_one_0.mlir.prj/aie_cdo_init.bin
Generating: /home/muhaawad/git/amd/iron/mlir-aie/programming_examples/basic/bug/add_one_0.mlir.prj/aie_cdo_enable.bin

$ aiecc.py --aie-generate-xclbin --aie-generate-npu-insts --no-compile-host --xclbin-name=test_1.xclbin --npu-insts-name=test_1.bin add_one_0.mlir

Generating: /home/muhaawad/git/amd/iron/mlir-aie/programming_examples/basic/bug/add_one_0.mlir.prj/aie_cdo_elfs.bin
Generating: /home/muhaawad/git/amd/iron/mlir-aie/programming_examples/basic/bug/add_one_0.mlir.prj/aie_cdo_init.bin
Generating: /home/muhaawad/git/amd/iron/mlir-aie/programming_examples/basic/bug/add_one_0.mlir.prj/aie_cdo_enable.bin

$ diff test_0.xclbin test_1.xclbin

Binary files test_0.xclbin and test_1.xclbin differ
$ diff test_0.bin test_1.bin
```

jgmelber, 2025-09-05:
Why does JIT show this verbose output too?

```

Generating: /home/muhaawad/git/amd/iron/mlir-aie/programming_examples/basic/bug/add_one_0.mlir.prj/aie_cdo_elfs.bin
Generating: /home/muhaawad/git/amd/iron/mlir-aie/programming_examples/basic/bug/add_one_0.mlir.prj/aie_cdo_init.bin
Generating: /home/muhaawad/git/amd/iron/mlir-aie/programming_examples/basic/bug/add_one_0.mlir.prj/aie_cdo_enable.bin

```

jgmelber, 2025-09-05:
> > Why does JIT show this verbose output too?
> > ```
> > 
> > Generating: /home/muhaawad/git/amd/iron/mlir-aie/programming_examples/basic/bug/add_one_0.mlir.prj/aie_cdo_elfs.bin
> > Generating: /home/muhaawad/git/amd/iron/mlir-aie/programming_examples/basic/bug/add_one_0.mlir.prj/aie_cdo_init.bin
> > Generating: /home/muhaawad/git/amd/iron/mlir-aie/programming_examples/basic/bug/add_one_0.mlir.prj/aie_cdo_enable.bin
> > ```
> 
> Comes from
> 
> [mlir-aie/lib/Targets/AIETargetCDODirect.cpp](https://github.com/Xilinx/mlir-aie/blob/90339e2ef66ac50a41ace5679f0fedc0f5e328eb/lib/Targets/AIETargetCDODirect.cpp#L63)
> 
> Line 63 in [90339e2](/Xilinx/mlir-aie/commit/90339e2ef66ac50a41ace5679f0fedc0f5e328eb)
> 
>  // TODO(newling): Get bootgen team to remove print statement in this function. 
> I have a hacky solution that redirects stdout temporarily in https://github.com/Xilinx/mlir-aie/tree/ypapadop-amd/redirect-stdout but it swallows error messages too :(

Maybe we should try bumping the bootgen submodule

ypapadop-amd, 2025-09-05:
> Maybe we should try bumping the bootgen submodule:
> 
> [Xilinx/bootgen@1d8a82e](https://github.com/Xilinx/bootgen/commit/1d8a82e19b1434d915b581fa80dd5d4da47ea2df)

Trying in https://github.com/Xilinx/mlir-aie/pull/2561

closed 2025-09-25 by AndraBisca