Xilinx/mlir-aie #2572: Make `IRON_CACHE_DIR` an environment variable

AndraBisca, 2025-09-10:
[This](https://github.com/Xilinx/mlir-aie/blob/main/python/iron/jit.py#L29) line in iron/jit.py sets the path for the JIT cache:
`IRON_CACHE_DIR = os.path.expanduser("~/.iron/cache")`

It would be nice to have the ability to change this without having to modify the code directly.

mawad-amd, 2025-09-26:
This was added in #2611. Any suggestions for where to document environment variables? Once documented, I will close this. @AndraBisca @jgmelber 

https://github.com/Xilinx/mlir-aie/blob/69ffd7a153de7eab5b397216057ba04bb420dad4/python/iron/jit.py#L26-L29

hunhoffe, 2025-11-10:
@mawad-amd To add my two cents, I think it's ok that this is an environment variable. As for documentation, I am thinking that we need to add an "IRON Configuration" section to the README and lists environment variables.

I don't see mentions of jit in the programming guide, so I don't think it needs to be added there unless we add a jit section there too. 

Thoughts @AndraBisca and @jgmelber ?

hunhoffe, 2025-12-29:
Configuration documentation was added and merged in (found here: https://github.com/Xilinx/mlir-aie/blob/main/programming_guide/iron_configuration.md).