Xilinx/mlir-aie #2586: Problem Implementing Broadcast with skip objectfifo for GRU network

iuwizeyimana, 2025-09-17:
I am currently implementing a GRU network on the NPU using the IRON API.

The hardware mapping used in this setup is shown in the picture [here](https://drive.google.com/file/d/1quabjDmI497TclCF4SYaLatUigAlu0JL/view?usp=sharing). 

For runtime data movement, I am following the pattern from the [single-core matrix-multiplication example](https://github.com/Xilinx/mlir-aie/blob/main/programming_examples/basic/matrix_multiplication/single_core/single_core.py), which serves as the baseline for my matrix multiplication compute tiles (A–F in the picture). No memory transformations are applied to the objectfifos transferring data between compute tiles.

The issue arises with compute tile **J**. As shown in the diagram, tile J participates in a broadcast with skip connection pattern as described [here](https://github.com/Xilinx/mlir-aie/tree/main/programming_guide/section-2/section-2b/02_Broadcast#broadcast-with-skip-connection). The tile consumes vector **H** from the mem tile a few cycles after other consumers of H, on which J depends for its second input (**Z**).

A FIFO of depth 2 between the mem tile and compute tiles works correctly in a non-tiled configuration (where N=n, M=m, K=k for inputs of size M×K and weights of size K×N), but fails in the tiled setup. Adjusting the FIFO depth such that each tile of the objectfifo that transfers H from the mem tile to the compute tiles has its own FIFO (as suggested in the broadcast-with-skip example) does not resolve the problem.

Interestingly, the issue persists as long as tile J is included as a consumer of the objectfifo even if it does not actually consume the vector in its core computation.

Any help with resolving this issue would be appreciated.

AndraBisca, 2025-09-25:
Hello @iuwizeyimana and thank you for submitting this detailed issue!
Could you please share your IRON design code so I may get a better understanding of how the ObjectFifos are set up?
Also when you say the design fails, do you get wrong results, or is there an error either during compilation or at runtime?

iuwizeyimana, 2025-09-25:
Thank you for the follow up!
I have condensed all the files needed to reproduce the error in [this repository](https://github.com/iuwizeyimana/GRU_on_NPU).

The IRON design is in the [gru_full.py](https://github.com/iuwizeyimana/GRU_on_NPU/blob/main/gru_full.py) file. 

By "design fails" I mean I am getting wrong results (all the output values are 0).

I found that adding a compute tile that reads the H values and store them in another tensor before feeding them to compute tile J fixed the error so I am thinking the error might be related to the size of the tensor that is being transferred.
However, I would not expect all the output values to be zero if that was the case, and it does not explain why just adding J to the [hmem2comp](https://github.com/iuwizeyimana/GRU_on_NPU/blob/c8111e1a747ff0aacaf075a673c1f6be4d13022b/gru_full.py#L270) objectFifo without using the values in the compute tile J would still result in all the output values being zero

closed 2026-05-13 by hunhoffe