Xilinx/mlir-aie #2597: Questios: Detailed device side runtime measurement

RonZ13, 2025-09-19:
Hi there,

I am wondering is there any method for measuring more detailed runtime? The example below shows the runtime collected from host side, but this includes the host to NPU data transfer time. Is there a way to measure each individual component of this total runtime:
- Host to NPU instruction transfer time.
- Host to NPU input data transfer time.
- Total NPU device runtime.
  - Kernel runtime.
  - Inter-tile data transfer time.
- NPU back to host time.

We would like to measure the impact of bad placement on device runtime. We believe that the runtime results gathered by the method below inlcude data transfer overhead between host CPU and NPU device.

https://github.com/Xilinx/mlir-aie/blob/c921519ebb67f4e575df59dcf6cabab0c8ee7a1d/programming_examples/basic/matrix_multiplication/test.cpp#L224-L233

We tried measure device runtime using hardware supported trace feature, but it is inconsistant and not aligned with the runtime measured from host side.

Regards,
Ron

andrej, 2025-09-22:
Thanks for your question.

You are correct that the host-side wall-clock measurement includes the data transfers.

As far as I am aware, trace would be the only way to separate out the individual components you are trying to measure. One thing to keep in mind with trace is that the numbers it reports are execution cycles. If you visualize it in Perfetto, it'll show as microseconds, but it's really cycles. Perhaps this is why the values compared to wall-clock time measured on the CPU seemed off to you; you'd have to multiply the cycles with the NPU's frequency to get a time.

Another limitation of trace is its limited buffer size, so you might only get a subset of the whole execution. For many purposes, that can be enough to identify where overheads are, but we understand it can be a bothersome technical limitation.

@asyms and team have been using trace to benchmark their kernels. They focused primarily on execution time of the core kernels. Perhaps they have additional useful input on how you can use trace effectively for runtime measurement purposes.

andrej, 2025-12-01:
Closing this due to inactivity. Feel free to reopen if more questions pop up.

closed 2025-12-01 by andrej